jakapoor/AMRUPT #2: Coherent detection necessary for phase estimation?

jakapoor, 2017-10-26:
In Dr. Kan's RFID-based system the interrogating transmitter and tag's RF oscillations are synchronized, but this is not the case with active tags. Is this a problem? I.e. there is no chance for coherent detection without carrier offset.

For details see section 3.2.4 of (Ma, Hui, Kan, 2016)

russellmsilva, 2017-11-07:
A possible solution to this problem would be connecting multiple external antennas to the CC1310 for solving an angle of arrival without the need for synchronizing multiple CC 1310's. Moreover, trilateration can be used in this step instead. The angle of arrival computations then can be used to calculate a position from multiple CC 1310's in order to triangulate a position (this information won't need to be synced to the level of nanoseconds because we will have already acquired it digitally).

I have already looked into the issue of acquiring IQ information from multiple external antennas. The following link provides some information to this task:
http://www.ti.com/lit/an/swra523b/swra523b.pdf

russellmsilva, 2017-11-07:
In one of Dr. Kan's systems, he utilizes beamforming on a 2-element antenna array to estimate the angle of arrival. This approach would also discard the need for nanosecond synchronization between receivers.

Details: Section II introduction in (Ma, Hui, Sharma, Kan, 2016)

jakapoor, 2017-11-08:
I've quickly looked over the document you provided, Russell, and I am still not convinced that antenna diversity helps us. 

If you look at Figure 1 in the document, and read the description of section 2.1 (Preamble-Based Antenna Diversity With External Switch), **it appears that this software patch is designed to get the CC1310 to switch between two antennas that are tuned to different frequencies.** If a sufficiently high preamble signal on antenna 1 is not heard within a certain time period, the CC1310 switches to antenna 2 and listens. At no point in the example are two frequencies / antennas being sampled simultaneously. In figure 7, the three receivers are different CC1310's, and they were set up for the purpose of comparing RF link quality between receivers tuned to only one antenna (1 and 2 for MCU 1 and 2), and another with antenna diversity. **In no location on this document could I find evidence that both antennas were being sampled simultaneously, which means our problem with achieving coherent detection remains.**

**Regarding your other point about synchronization, we must (at all costs) avoid the need for tight temporal synchronization between antenna arrays (our ground nodes).** Luckily, none of the designs we've been considering seem to rely upon that, but the story is different within the arrays: the within-array antennas must be sampled simultaneously using a common clock and a common oscillator in order to achieve coherent detection. **That said, the paper I included in my first message (Chen et. al. 2012) has found an intriguing potential solution to the problem of un-synchronized radios in a phase-based direct-finding application using known-location radio beacons.**

In my somewhat naive opinion, I feel like we're approaching the end of the line with using the CC1310 for coherent detection, unless we can get multiple chips synchronized with a local oscillator or with a known-location beacon. 

We need to figure this out quickly, so we can move over to our plan B architecture (separate demodulators, mixers, and a multi-channel ADC driven by an LO) if that's the only viable option.

jakapoor, 2017-11-09:
RESPONSE BY EMAIL FROM RUSSELL
========================
Julian Kapoor,

I read through your email, and I somewhat agree on your position. I agree on your remark about the lack of evidence for simultaneous sampling in the document (http://www.ti.com/lit/an/swra523b/swra523b.pdf). In my opinion, I consider the document as something that may provide some of the groundwork for connecting two external antennas. I do not view the document as providing a design completely in parallel with the components and qualities that will be necessary in our system. Their flowchart/design based on the "Preamble Quality Threshold" algorithm that they present is unrelated to our objective. However, I believe the code they present can be manipulated, so that this algorithm is not implemented and the antenna switching does not occur. I cannot find the part in the document where the antennas are tuned to two different frequencies. Since both antennas were designed to receive the same transceiving packets, I do not see why they would set the receivers to two different constant valued frequencies. The frequency offset mentioned later in the document seems to be altered in order to experiment with the power levels of the radio signal. Moreover, this frequency offset seems to be a variable quantity that can be set to zero, but I may have misread some of the graphs.

On another note, I would not immediately "jump ship" to our plan B architecture, as I believe getting in contact with Texas Instruments can provide us vital details that we might have missed in searching through various online forums and resources. 

I would still like to meet at 2 PM on Thursday to draft our contact with Texas Instruments and any other individuals of relevance.

Best,
Russell

jakapoor, 2017-11-09:
> Since both antennas were designed to receive the same transceiving packets, I do not see why they would set the receivers to two different constant valued frequencies.

I believe this is to accommodate for issues with one of the two channels having interference.

jakapoor, 2017-11-18:
I'm going to conclude that the answer to this question (i.e. is coherence required between SDRs) is yes with a but, or no with a maybe. 

Essentially, we need to drive multiple CC1310's with a single oscillator. If we can do this, and both the RF oscillations and sampling from the ADC are driven by a PLL (and therefore all oscillations are some multiple of the LO oscillations), then we just need to find a way to calibrate for small offsets between the radios.

I'm going to close this issue. We have not found a solution yet, but we essentially have our answer: **Coherent detection (_ish_) is necessary.**

jakapoor, 2017-11-18:
I'm going to conclude that the answer to this question (i.e. is coherence required between SDRs) is yes with a but, or no with a maybe.

Essentially, we need to drive multiple CC1310's with a single oscillator. If we can do this, and both the RF oscillations and sampling from the ADC are driven by a PLL (and therefore all oscillations are some multiple of the LO oscillations), then we just need to find a way to calibrate for small offsets between the radios.

I'm going to close this issue. We have not found a solution yet, but we essentially have our answer: Coherent detection (ish) is necessary.

closed 2018-02-09 by jakapoor