jakapoor/AMRUPT #11: How to achieve coherent detection?

jakapoor, 2018-02-28:
Finally, in our AMRUPT meeting yesterday (02/27/2018), I mentioned that sample rate may be an important part of the choice to pursue our intended single-radio receiver approach to direction finding. I also wondered why this one-radio approach doesn't seem to be more common in others' DF solutions. I briefly forgot why I brought that up, and the group came to the consensus that sample rate was not more of a problem in a single radio approach than a synchronized multiple radio approach. 

In thinking more about this I now recall why I brought this up: **In order to compare phase differences on the multiple channels we'll need a high enough sample rate to reconstruct the original RF waveform.** That might not be a necessity with multiple radios, and almost certainly is not a necessity with the mixer/demodulator approach (since we'd be working with a much lower [intermediate] frequency that could be sampled in the kHz range). If my signal analysis theory serves me right, we will need to sample the RF signal at at least twice the highest frequency we are going to be using in our RF transmissions. Assuming that is correct, that means **we'd have to sample each antenna's output at over 600 MHz if we go up to 300 MHz (!)**, and I highly doubt the CC1310 can handle obtaining I/Q samples at that rate. I'd appreciate it if someone can correct me about this logic.

jakapoor, 2018-02-28:
#(Response from Ian Briggs by email - 02/27/2018)

"That's a good point.  If we are going with the RF switch approach, I'm not sure how a mixer/demodulator would handle the sudden switch from one antenna to the other.  The output end of a mixer might look bizarre immediately after the switch until it settles back down to the usual sinusoid waveform.  We might be able to get away with using a mixer/demodulator in the setup if we assume that using a mixer won't add a significant time delay/phase shift to the sinusoid when the switch is activated (or if this time delay/phase shift is consistent and can easily be accounted for).  In other words, we wouldn't be able to use the phase of the waveform immediately after the shift, but we might be able to wait until it settles down and predict what the sinusoid from the first antenna would look like to make a phase difference measurement.  It's probably worthwhile asking Professor Kan about this if sampling the RF band directly is not an option."

jakapoor, 2018-02-28:
I haven't thought far enough ahead about the possibility of combining the mixer/demodulator and RF switch approaches, but that seems to me like a separate downstream issue.

I very much agree though, that it's worthwhile asking Dr. Kan about whether it's possible to sample the RF band directly. It _might_ be possible, but 1) we likely wouldn't be able to do this with the CC1310, and there might be really high quality ADCs that have a sufficiently high sampling rate, but 2) those ADCs are prohibitively expensive, and 3) sampling rates that high often suffer in having crummy bit depth, which would lead to high levels of quantization error.

**If anyone has reading materials that deal with this issue please let everyone know, and add the relevant PDF(s) to GitHub in the '\Literature\General_radio_direction_finding\Coherent_detection_w_SDRs' folder. I'll do the same.**

Any other thoughts?

russellmsilva, 2018-03-01:
For ideal sinusoidal signals, the CC1310 does not have to sample antenna outputs at twice the maximum frequency of the transmitted signals. Later in this response, I will go over cases in which the under-sampled signal may pose a problem in the nonideal case. 

Assuming our sampling rate is constant, sampling a sinusoidal wave lower than the Nyquist frequency will produce an equivalent reconstructed signal. This reconstructed signal will have the same wave pattern and structure of the original signal. The only difference is that this reconstructed wave will be at a lower frequency. The above holds true for infinitely slow sampling rates (stationary sources) .

If we take the phase difference between two sinusoidal waves that have been reconstructed from a lower sampling frequency than the Nyquist at any given time, their phase differences at that given time will be the same as if we were using the original sinusoidal function. Since diagrams are a lot better at explaining this, I will leave that to the revised proposal!

As for Ian's suggestion, RF switches in the market today work at switching speeds around 100 nanoseconds. Because RF switching speeds are so fast, our under-sampling problem may even negate the bizarre effects that would occur on the reconstructed signal during the RF switching. Additionally, we can program the Raspberry Pi to stop collecting data for a couple of milliseconds to completely eradicate any error from RF switching speeds. 

In addition, I do not believe the RF switching process would add a time delay/phase shift to the next antenna channel it will be switching to. Once the RF switch is locked onto a channel, the signal should be in real-time and will be separate from signals on other antenna channels.

As with sources of error from undersampling:

1) We cannot use frequency modulated signals on radio tags. We will have to make sure that the transmitted signals represent sinusoidal waves of constant frequency in order for undersampling to work. 

2) Noise will pose more of a problem in undersampling than oversampling. However, it is important to note that noise causes as much of a problem in the synchronized local oscillator system as it does in the RF switch system. If the synchronized local oscillator system has the same lower-than-Nyquist sampling rate (we proposed using CC1310s for this system last semester), it will be just as susceptible to noise problems (this can be discussed at future meeting). 

The CC1310's I/Q values should come from a signal that has gone through many noise filtering components. If environmental noise/interference should pose a significant problem, probabilistic measures such as MLE or MAP can be integrated into the signal reconstruction, or the Raspberry Pi could increase listening time (to one antenna channel) if the reconstruction is generating an irregular signal at a given time.

Sources:
Sampling lower than the Nyquist: 
http://www.wescottdesign.com/articles/Sampling/sampling.pdf (Page 14)
RF Switch Basics:
https://www.pasternack.com/t-basics-of-rf-switches.aspx

jakapoor, 2018-03-01:
Awesome! Thank you Russell! 

This is very clear, and greatly encouraging to me that we're still on the right track. If anyone disagrees with any of Russell's conclusions, please speak up! 

A few follow-up points:

> We cannot use frequency modulated signals on radio tags. We will have to make sure that the transmitted signals represent sinusoidal waves of constant frequency in order for undersampling to work.

Good to know, and important for considerations of forward compatibility for the multi-frequency (MF) version of the system. That said, since we're considering the Heuristic Multi-frequency Constant Wave (HMFCW) approach to a MF system (sensu Ma et al 2016), which uses several sinusoidal waves of different frequencies delivered in series, we should still be OK here, correct?

> Noise will pose more of a problem in undersampling than oversampling.

I'll admit this worries me. This system will be operating not in a very noisy (from other RF sources) environment but a very cluttered environment, resulting in multipath signals that will effectively add noise. To me, the effects of the degree of undersampling on phase difference estimation seems a legitimate thing to model.

> However, it is important to note that noise causes as much of a problem in the synchronized local oscillator system as it does in the RF switch system.

Here I _think_ I disagree. Noise, by its nature, is random and time varying. Since the RF switch technique involves the sampling of different channels at different times the noise experienced by antenna 1 at time 1 will be different from the noise experienced by antenna 2 at time 2. In the case of a synchronized system, however, both channels are sampled simultaneously (even if undersampled), and are therefore more likely to experience the same noise characteristics. I believe a number of op-amps specifically capitalize upon this fact to allow for very high gain with low noise susceptibility. This _might_ therefore be a very important consideration for how robust the system is to RF noise.

> ...probabilistic measures such as MLE or MAP can be integrated into the signal reconstruction...

Just to make sure we don't misinterpret acronyms: Maximum Likelihood Estimation (MLE) and Maximum A Posteriori (MAP). I'd be interested in hearing more about this at some point.

> ...or the Raspberry Pi could increase listening time (to one antenna channel) if the reconstruction is generating an irregular signal at a given time.

I don't think this will be possible given that the mobile nodes (Tx) will not know that they need to increase transmission time, and we certainly don't want the tags transmitting signals for a significantly longer duration than they're typically going to be listened to by ground-nodes (for energy consumption reasons). Thoughts?

Finally, I have found on Web of Science a series of papers using SDRs and RF switches in localization applications. Although I suspect the rest of the team may already have come across these, check them out if you haven't seen them yet (Literature section of GitHub). Here's a hardware block diagram from one of them (He et al 2015):

[image]

Although these papers appear to use the RF switching technique to compare inputs from two antennas on a single radio, they appear to use a very different approach to arriving at phase differences from what I understand of the current AMRUPT approach (i.e. wave reconstruction): analysis of the harmonic content of the resulting power spectra of the combined signal.

That's all from me for now. Thanks!

jakapoor, 2018-03-29:
# Update
Our hardware approach has changed to include multiple RTL-SDRs being driven by a common local oscillator. Therefore, the question now is what calibration techniques are required to make coherent detection possible. Both Whiting and Krueger have published work on doing this with GNU radio blocks (some of them custom blocks).

jakapoor, 2018-06-13:
As the current issue about achieving within-array coherent detection is no longer a matter of hardware approach, but now an issue of software, I am closing this issue.

closed 2018-06-13 by jakapoor