jakapoor/AMRUPT #15: Access to Radio Tags that Transmit Unmodulated Vertically-Polarized Sinusoidal Waves

russellmsilva, 2018-06-15:
In AoA and PDOA approaches, it will be critical to have radio tags transmit unmodulated sinusoidal waves in order to correctly determine the phase shifts of an impinging received signal between antenna elements in an array. Also, our receivers are connected to antennas that are linearly-polarized, so having radio tags that have linearly-polarized antennas will allow for greater downlink (transmitter to receiver) propagation distances. Is it possible to have access to these tags when testing the software/hardware of the receiver network? It is unclear whether the continuous TX option on the CC1310 (SmartRF Studio) supports this, and I am unsure on whether this can be done in TI-RTOS.

jakapoor, 2018-06-16:
>In AoA and PDOA approaches, it will be critical for radio tags to transmit unmodulated sinusoidal waves in order to correctly determine the phase shifts of an impinging received signal between antenna elements in an array.

Please see my comments on #12 with respect to the need for unmodulated sinusoidal transmissions. I believe these will not be compatible with our system. Whether we end up with an UWB signal or not, some degree of modulation will be necessary.

>Also, receivers are connected to antennas that are linearly-polarized, so having radio tags that have linearly-polarized antennas will allow for greater downlink (transmitter to receiver) propagation distances.

We will be operating with whip-antennas on the transmitters, which will produce signals that are similar to the reception pattern of the whip antennas of the receivers. If reception range becomes a major issue we can revisit this.

>Is it possible to have access to these tags when testing the software/hardware of the receiver network? 

I think it best that we start out with the CC1310 launchpad configured in transmit mode. The issues concerning detection range seem to me far less important right now than the issue of coherent detection and system calibration.

>It is unclear whether the continuous TX option on the CC1310 (SmartRF Studio) supports this, and I am unsure on whether this can be done in TI-RTOS.

I am unclear what "this" refers to. Do you mean linear polarization of signals? That is related to the antenna design, not to the RF front-end I believe. Could you please clarify your question?

russellmsilva, 2018-06-16:
> Please see my comments on #12 with respect to the need for unmodulated sinusoidal transmissions. I believe these will not be compatible with our system. Whether we end up with an UWB signal or not, some degree of modulation will be necessary.

If we implement TDOA, these transmissions will be incompatible. However, subspace angle of arrival and all other interferometry methods require signals of constant phase fluctuation (the degree to which phase changes overtime in a signal is constant). Frequency modulation in a signal would prevent constant phase fluctuation from occurring. Perhaps there are some interferometry systems that use signals of changing phase fluctuation; however, sending and processing these signals would add an unnecessary layer of complexity to our system.

> We will be operating with whip-antennas on the transmitters, which will produce signals that are similar to the reception pattern of the whip antennas of the receivers. If reception range becomes a major issue we can revisit this.

The antennas that are currently hooked up to our RTL SDR receivers are linearly polarized antennas.

> I think it best that we start out with the CC1310 launchpad configured in transmit mode. The issues concerning detection range seem to me far less important right now than the issue of coherent detection and system calibration.

I agree, I am unsure how to get unmodulated continuous sinusoidal signals transmitting from the CC1310.  I have looked through a couple of online forums, and I am still trying to figure out how to accomplish this simple task with the CC1310. 

> I am unclear what "this" refers to. Do you mean linear polarization of signals? That is related to the antenna design, not to the RF front-end I believe. Could you please clarify your question?

"this": unmodulated continuous sinusoidal signal transmissions

jakapoor, 2018-06-17:
Hi Russell, see my comments below:

> If we implement TDOA, these transmissions will be incompatible. However, subspace angle of arrival and all other interferometry methods require signals of constant phase fluctuation (the degree to [...]

I still strongly disagree. Can you provide me with a reference or two indicating that an unmodulated sinusoidal carrier is necessary for both PDOA and AOA? Hopefully we're agreed that an UWB PRN signal is best for TDOA, but I also believe it is compatible with (possibly even necessary for) PDOA, and compatible with AOA. 

Although I understand the concept that a constant phase change over time is _helpful_ for AOA approaches, I don't believe it is _vital_ for AOA and I believe it is insufficient for PDOA. To achieve a constant phase change over time you need a signal with a stable frequency (i.e. a sinusoidal carrier, as you suggest). That said, an UWB PRN signal generated with OOK (which uses a sinusiodal carrier) still contains a series of digital "ones" (as opposed to zeros) that contain pure sine waves of a known frequency. Any segment of the UWB signal that is represented by a digital 1 (and therefore contains a known frequency sinusoidal carrier wave) should be sufficient for both AOA and PDOA. And even if we don't use OOK, any other form of modulation (frequency, for instance) can be demodulated to extract the original carrier's pure tone information. Furthermore, the advantages of the UWB signal will allow us to be sure that what we are aligned to the same time point in the signal when we measure the received phase difference of the carrier (this is basically the flip side of TDOA, where we align the signal and look for the time difference). With an unmodulated signal there will be no way to accurately align the signals between distributed ground-nodes, which would be important for TDOA, and equally for PDOA, though, admittedly, unnecessary for AOA since the received signal will be coherently detected, and the poor temporal resolution of an unmodulated carrier would be unimportant. Even if the UWB PRN code were not compatible with AOA, a short sinusoidal preamble before that UWB signal could help us to achieve both goals (AOA, then PDOA refined by TDOA). Please let me know if you disagree still, and if so, try to find a paper we can use to figure out where our logic is failing us. I'll do the same. Incidentally, if you recall, Dr. Kan mentioned PRN codes in the context of PDOA, so he may be a good resource to run this issue by.

>I agree, I am unsure how to get unmodulated continuous sinusoidal signals transmitting from the CC1310.

I am aware that there is a CW (constant wave) functionality, though I don't recall where it is. I'll look into this. Couldn't you configure the radio in OOK mode and feed it only a long string of ones? That ought to produce an unmodulated sinewave. Although, due to the above argument, I would ordinarily try to dissuade you from spending time on this (since I still believe we need a modulated signal), I actually think we need to figure out how to produce an unbroken sinewave so that we can try to generate an UWB signal by toggling the RF core's power (a la Kruger). For that reason I think this is a worthy pursuit for now.

Thanks!

russellmsilva, 2018-06-17:
> I still strongly disagree. Can you provide me with a reference or two indicating that an unmodulated sinusoidal carrier is necessary for both PDOA and AOA? Hopefully we're agreed that an UWB PRN [...]

I think that we are on the same page for realizing that we need to acquire a signal with a stable frequency (i.e. a sinusoidal carrier) for AoA and PDOA. Although I did not foresee it, utilizing OOK (or another modulation scheme) allows a receiver to obtain a pure sinusoidal tone from a UWB signal.

> Any segment of the UWB signal that is represented by a digital 1 (and therefore contains a known frequency sinusoidal carrier wave) should be sufficient for both AOA and PDOA. And even if we don't [...]

From Kruger's Dissertation: "In summary, the OOK-modulated signal can be viewed as the linear combination of a narrowband signal with high spectral density and a wideband signal with low spectral density. For a random code, half of the code bits will be ones. Consequently, about half of the signal’s energy will be contained in the narrowband main lobe of the unmodulated carrier component, while the other half is spread over a wide range of frequencies."

Great idea, I have not seen this type of implementation stated explicitly in any of the literature I have read through before. Can you reference a paper on this approach if you have one? I will try to consult with Dr. Kan on this. The only skepticism I have is that the lower spectral power density of frequencies surrounding the narrowband sinusoidal carrier could constructively or destructively interfere with the narrowband sinusoid (correct me if I am wrong), creating frequency instability in the narrowband sinusoid. 

>Furthermore, the advantages of the UWB signal will allow us to be sure that what we are aligned to the same time point in the signal when we measure the received phase difference of the carrier (this is basically the flip side of TDOA, where we align the signal and look for the time difference). 

I am having trouble understanding this, can you elaborate further?

>With an unmodulated signal there will be no way to accurately align the signals between distributed ground-nodes, which would be important for TDOA, and equally for PDOA

I am a little skeptical here. In Kruger's dissertation, beacons assist in "estimating the difference in arrival time without accurate estimates of the sample time, with unsynchronised real-time clocks, and with differences in the receiver's sample rates." However, it is also stated that Sample-of-Arrival (SOA) values (determined from modulated DSSS signals) assist in this task. I still see a major flaw in utilizing gold codes in dense multipath environments (#12), and I believe that the use of beacons could allow for coherent detection among distributed ground nodes in a PDOA/TDOA system without SOA values. 

>Even if the UWB PRN code were not compatible with AOA, a short sinusoidal preamble before that UWB signal could help us to achieve both goals (AOA, then PDOA refined by TDOA).

Good idea. It is important to note that most literature on subspace techniques for AoA (e.g. MUSIC) assume the use of narrowband signal sources when forming a model based on received signals impinging on an antenna array such as [here](http://www.girdsystems.com/pdf/GIRD_Systems_Intro_to_MUSIC_ESPRIT.pdf) and [here](https://pdfs.semanticscholar.org/7215/ae3db824d42d2b9f32077e5908a0fb97a65e.pdf). There are a few methods out there that extend MUSIC's capability to broadband signals including [here](https://pdfs.semanticscholar.org/bb36/19c8a498394593318919b6ac869bc32cc19b.pdf) and [here](https://smartech.gatech.edu/bitstream/handle/1853/5110/yoon_yeosun_200407_phd.pdf).

jakapoor, 2018-06-18:
>I think that we are on the same page for realizing that we need to acquire a signal with a stable frequency (i.e. a sinusoidal carrier) for AoA and PDOA. 

My intuition agrees, but we really should confirm this.

>Although I did not foresee it, utilizing OOK (or another modulation scheme) allows a receiver to obtain a pure sinusoidal tone from a UWB signal.

This is something we discussed in an AMRUPT meeting several months back (the one attended by Dr. Kan on March 27th, 2018) during our initial discussion as we first started thinking about the use of CDMA scheduling. Dr. Kan was recommending UWB PRN codes to solve issues of tag scheduling, rather than specifically to enable AOA, PDOA, or TDOA approaches. We later discussed, however, the advantages of having half the spectral content at the carrier frequency and the other half spread over a wide range of frequencies for specifically the purpose of combining TDOA and AOA/PDOA. Does that ring a bell? You may want to refer to your notes for the March 27th meeting. 

>Can you reference a paper on this approach [i.e. OOK with UWB signal in PDOA/TDOA/AOA DF] if you have one?

Unfortunately, Krueger's work is the only work I have found that references the advantages of the narrowband and broadband content of OOK. I'll keep looking though.

>The only skepticism I have is that the lower spectral power density of frequencies surrounding the narrowband sinusoidal carrier could constructively or destructively interfere with the narrowband sinusoid (correct me if I am wrong), creating frequency instability in the narrowband sinusoid.

I would not have thought this would be an issue, but I am certainly not an authority. The content of this signal can be thought of in two ways, in the frequency domain and in the time domain. In the former, the signal will have a very broad peak with a higher very narrow peak superimposed at the carrier frequency; that doesn't help us all that much. In the latter (i.e. time) domain the signal will look like a series of short  sinusoidal carrier pulses turning on and off (pseudo randomly) at a rate of 1 MHz. In the time domain it seems like there shouldn't be an issue with constructive/destructive interference. Also, in the frequency domain, the power contained in any one "bin" of frequencies other than the carrier itself will be very very small compared to the carrier. To me that means there's not a great deal of risk that the broadband spectral content will affect the carrier. Again, I really don't think this will be an issue, but it's certainly worthy of further attention.

> >[Julian] Furthermore, the advantages of the UWB signal will allow us to be sure that what we are aligned to the same time point in the signal when we measure the received phase difference of the carrier (this is basically the flip side of TDOA, where we align the signal and look for the time difference).

>[Russell] I am having trouble understanding this, can you elaborate further?

UWB PRN (Gold) codes are like a long morse-code series of pulses, resembling a random sequence, but specifically selected to have low self-correlation when not perfectly aligned. That means that Gold codes are ideal for cross correlation, where the template is matched with the received signal. This is because, ideally, there will be a strong peak when you have correctly aligned the template and no peak when the signals are misaligned. In the case of TDOA, you are sampling at a pair of receivers, and for simplicity, we're assuming that the receivers are coherent (i.e. sample 1 of receiver 1 is exactly coincident with sample 1 of receiver 2). When the same Gold code is received at both receivers, you cross correlate them with the template until you get the peaks representing a match with the template. The number of samples you had to shift the signal to match the template (with respect to the other receiver's signal) represents the difference in time of arrival of the signal. In the case that the Gold code was transmitted equidistant from the two receivers the number of samples that needed to be shifted for the signals at receiver one and two should be equal. 

Now, in the case of PDOA (and this is where I am guessing, since I haven't actually read about this explicitly) I believe what is happening is very similar: the two received signals are cross correlated with the template, and the exact shift in time (representing TDOA) is found. But now, instead of using that time shift information in TDOA you use those two aligned signals (signal 1a at receiver 1 and signal 1b at receiver 2) and you zoom in to the sinusoidal content of one (or more, perhaps all?) of the digital ones in the PRN code. Zoomed in to the sinusoidal content, and with your knowledge of the frequency of the carrier, you can then identify the difference in phase of the carrier of the two signals. **This is where I can't help, and we need further clarification: I'm still fuzzy on whether you'd actually compare the signals' carriers' phase differences when the signals had been aligned through cross-correlation, or whether you compare the phase shift of the signals aligned in time (i.e. the signals will not be overlapping). This, I think, is a major missing element in both of our understanding of how PDOA works.** 

> I am a little skeptical here. In Kruger's dissertation, beacons assist in "estimating the difference in arrival time without accurate estimates of the sample time, with unsynchronised real-time [...] [... cut ...]
**Absolutely! This is a very important conclusion, and one which I agree with.** The purpose of this discussion, as I have reiterated _ad nauseum_ :) during our AMRUPT meetings is to make sure we understand what we will eventually need this system to do, so that we don't have to completely redesign our system down the road. We need to make sure that the hardware we're using is compatible with this direction, and this is why I've been so insistent on the above content making its way into the "Forward compatibility" section of the proposal.

russellmsilva, 2018-06-18:
Thank you for the detailed response!

> Regarding the first issue, yes, beacons help to align non-coherent distributed receivers in Krueger's system, and indeed these beacons are essential for this reason. Regarding the second issue, the [...]

Beacons can help improve accuracy in a system and enable the synchronization of real time clocks among multiple ground node receivers without PRN codes . Please see [here](https://github.com/jakapoor/AMRUPT/blob/master/Literature/General_radio_direction_finding/Types/TOA_TDOA/Weiser%20et%20al._2016.pdf).

> The purpose of this discussion, as I have reiterated ad nauseum :) during our AMRUPT meetings is to make sure we understand what we will eventually need this system to do, so that we don't have to [...]

Thank you for your reiteration. However, the use of PRN codes with a modulation scheme (On-Off Keying) under a dense multipath environment could devastate measurement accuracy in a TDOA/PDOA system, because multiple copies of the PRN code will arrive at receivers within the PRN sequence’s period. For example, consider a ground node localization system utilizing PRN codes similar to the ones used in a GPS satellite, which have a period of 1023 chips corresponding to [1 millisecond](http://read.pudn.com/downloads85/ebook/326017/Fundamentals%20of%20Global%20Positioning%20System%20Receivers/booktext05.pdf). If a receiver is 100 meters from a radio tag, and a scattered signal must travel 50 more meters than the LOS signal, the scattered signal will reach the receiver 0.167 microseconds after the LOS signal (50 meters/speed of light). This means that cross correlating the LOS with a local PRN code would be disrupted or delayed. Furthermore, it more likely for scattered signals to enter within the PRN code's period than after it. This is because a scattered signal would have to travel at least ~280 miles in order to arrive at a receiver after the 1.5 millisecond PRN signal.

jakapoor, 2018-06-19:
# Regarding your previous two posts about PRN codes in multipath environments:

Your latter comment signal scattering will not be a major issue is correct (as far as my understanding goes), though the reasoning you include is only part of the story. I've explained this in previous issues, but to recap those comments: there is low autocorrelation between a PRN code and itself at nonzero lags, and even if a delayed signal of relatively high amplitude significantly overlaps the LOS signal that original signal will likely be recoverable. If the reflection is shifted an exact integer number of chips over, this might impede the recovery of the signal, but it may still be recoverable even under these conditions. I can find the papers I have read that convince me of this, but generally, this is exactly why PRN codes are so useful. One does not need to stagger them. They (as part of a class of CDMA [code division multiple access] signals) are an alternative to TDMA (time division multiple access). The 1.5 ms signal spacing you referred to in your original posting above is a form of TDMA. So, I agree with your conclusion, but not only for the reason you mention.

# Regarding narrowband transmissions

>Dr. Kan advised me not to use UWB signals in our system, as they would complicate the design of receivers. 

We have to determine ourselves whether the additional complication is warranted. The decision comes down to whether it is necessary in order to achieve our design objectives, not whether it is complicated.

> ... the systems developed by Dr. Kan use PRN injected narrowband continuous waves (CW).

I think we may be using terminology incorrectly then. I had assumed that the combination of a narrowband continuous wave and PRN modulation (using OOK) would create a signal that could be called UWB. The spectral content would certainly be broadband when viewed in the frequency domain. So this may simply be an issue of semantics. To me it sounds like the type of broadband signal I was considering for a TDOA approach would also be compatible with PDOA if this kind of signal is acceptable (i.e. the one used by Krueger in his "Thrifty" TDOA system).

>He [Dr. Kan] assured that triangulation with MUSIC should obtain at least 5 meter accuracy for the above specifications, as long as signals transmitted from radio tags have enough power.

I believe that this is an overly optimistic take on the situation and that we need to be prepared for the possibility that this will not be the case. Work by MacCurdy and Richardson here at Cornell suggested that a completely AOA-based triangulation system will be woefully inadequate to achieve high resolution. Recall also that our 5m target is only the first step, and that we are trying to achieve the maximum resolution possible given the constraints of our hardware and budget.

> For these reasons, I will continue to put most of my focus on an AoA implementation that is forward compatible with a narrowband CW PDOA implementation in the proposal. Please let me know if you agree with me that this is the right direction.

The AOA-based approach is definitely how we want to proceed (continuing with phase one), but we need to be clear about what "narrowband CW PDOA" actually means. We still need to nail down where the PRN codes fit in. The reason that this is important now is that it impacts our transmitter design (we know we can produce a CW, but can we produce a PRN code?).

jakapoor, 2018-06-21:
# Update

I am closing this issue. We have essentially determined that we want to be able to modulate a basic carrier, potentially with a PRN code using OOK, which means that the initial inquiry, "Access to unmodulated linearly-polarized sinusoidal wave-producing transmitters" is no longer relevant. 

The tags we are using are the CC1310 transceiver, which can produce unmodulated CW signals (how is another question). And these tags support OOK modulation, though not at the rates we might need if we use PRN codes (ca. 1MHz).

The discussion in this issue has expanded into a more general discussion about the costs and benefits of narrowband and broadband signals, so the material may still be useful for review going forward.

In the future we should take care not to let the discussion drift from the original question, and to create new issues to discuss such content.

closed 2018-06-21 by jakapoor