jakapoor/AMRUPT #24: Angle of Arrival System Testing and Improvements

russellmsilva, 2018-08-17:
The first tests were performed with the coherent receiver placed in the middle of a baseball field. Preliminary angle of arrival results resulted in unstable sporadic angles. It was found shortly after that there was an RF signal in the environment transmitting at 431 MHz (we were using 433 MHz). It is debatable whether this would be an issue with the small receiving bandwidth being used (~1 KHz).

I was not able to test the MUSIC/root MUSIC algorithms sufficiently outside because the power supply only lasted one hour (with the signal source and low-battery laptop connected). Further tests were made indoors in Duffield. For the root MUSIC/MUSIC algorithms, the resulting angle took ~5 seconds to stabilize once the transmitter was put in a fixed position (see improvement 1). 433 MHz transmissions resulted to NULL values from the root MUSIC algorithm; however, 866 MHz transmissions never resulted in a NULL value (see improvement 2).

866 MHz transmission tests:
Indoor multi-path effects (resulting in a clearly unexpected angle) resulted when the transmitter was pointed more than 30 degrees away from the line of sight. Towards the end of testing last night, stable angles within 10 degrees of the actual angle of arrival (measured by protractor/string) were observed. These last observations were premature though, and further testing must be done to see if these were fluke values.

Improvements to be made:

1. There seems to be an instability issue with the GUI Compass. After some research, there seems to be a problem with it as shown here and here. Therefore, the following tests will be done using the console or a GUI float sink (displaying the float value result of MUSIC/root MUSIC on the top block GNU Radio GUI).

2. The mast contained a platform (for the coherent receiver, Raspberry Pi, and usb hub) within the near field of antenna array elements. The platform was 2 feet from the antenna array (near field is one wavelength - ~2.3 feet for 433 MHz). This platform will be placed further from the near field to reduce EM interference with electronic devices. 

3. No matter the frequency, gain, or bandwidth setting, RTL SDR 1 receives the greatest sinusoidal amplitude, RTL SDR 2 receives a lower sinusoidal amplitude, RTL SDR 3 receives an even lower sinusoidal amplitude, and RTL SDR 4 receives the lowest sinusoidal amplitude. Please see this video. The RF gain of each receiver had to be increased for the fourth RTL SDR to receive enough power. This phenomenon may adversely effect beamforming (as implemented in Sam Whiting's system). Since receiver 1 always receives a signal of much greater power than receiver 4, the MUSIC/root-MUSIC algorithms may be adversely effected as well.

Please feel free to shoot me any suggestions or ideas. A next set of tests will be performed after the above improvements have been made or reconciled.

jakapoor, 2018-08-17:
Thanks for this detailed information Russell. I'll respond to the specifics below, but first I want to make a few general comments / suggestions.

1) Some of this information will be redundant with what you've included above, but we need a centralized location to compare future empirical tests of the performance of the system. Can you upload to GitHub a document (or documents) detailing a) your goals for this empirical test (i.e. what - specifically - did you want to test? b) the setup (i.e. what hardware did you use, where, and exactly when, who was present, any other relevant notes), c) the parameters of the test(s) (i.e. what settings were you using, what versions of what code, _specifically_ referencing where they are in GtHub), d) what specific tests you conducted, and e) the results (I mean data here, not only general descriptions) along with your interpretation of what those results mean. **This is absolutely essential in order for us to actually benefit from these tests: a poorly documented test essentially never happened.**

OK, now to the specifics:

> The first tests were performed with the coherent receiver placed in the middle of a baseball field. Preliminary angle of arrival tests with 2 antenna elements resulted in unstable sporadic angles. It was found shortly after that there was an RF signal in the environment transmitting at 431 MHz (we were using 433 MHz). It is debatable whether this would be an issue with the small receiving bandwidth being used (~1 KHz).

Explain the hardware setup. How high was the rig, were you on mains power? Also, even preliminary tests should be clearly documented. How sporadic were these two-antenna-element angles? What DF algorithm were you using? Remember to include the result data on the repository. How do you know there was significant noise? How did you take these measurements? 

> I was not able to test the MUSIC/root MUSIC algorithms sufficiently outside because the power supply only lasted one hour (with a signal generator and low-battery laptop connected). Further tests were made indoors in Duffield. 

The low battery capacity is another reason to plan out exactly what you are going to do out there. Let's come up with a plan together before your next test. Please make sure to document your indoor tests fully as well. 

>For the root MUSIC/MUSIC algorithm tests indoors, the resulting angle took ~5 seconds to stabilize once the transmitter was put in a fixed position (see improvement 1). 

Thoughts about why this might be the case? This is definitely not something we can tolerate in a wildlife system, where animals are in place only momentarily. We'll need to understand why that's happening. Could it be related to the snapshot length required for MUSIC? 

>433 MHz transmissions resulted to NULL values from the root MUSIC algorithm; however, 866 MHz transmissions never resulted in a NULL value (see improvement 2).

Can you explain this? What do you think might be happening? We ultimately need to be able to work with 433 MHz and down in frequency. Transmission properties (attenuation due to vegetation) at 866 MHz aren't compatible with a wildlife system.

>Indoor multi-path effects (resulting in a clearly unexpected angle) resulted when the transmitter was pointed more than 30 degrees away from the line of sight. Towards the end of testing last night, stable angles within 10 degrees of the actual angle of arrival (measured by protractor/string) were observed. These last observations are premature though, and further testing must be done to see if these were fluke values.

First, how do you know this was the effect of multi-path interference, versus some other issue? Also, how can a non-directional transmitter be "pointed" in any particular direction? Do you mean that the transmitter was located 30 degrees with respect to the array? Please provide the data for this test. Why are these observations premature? Remember that data collection should be done systematically; you need to have a plan for exactly what information you want, how you will record it, and what you will do with it when you're done. This all comes back to the part of the project proposal that was to be devoted to creation of standardized test protocols.

Regarding improvement # 1:
Is there a numeric output from the GUI Compass? How will that information be logged for later analysis? We want to avoid any error due to inaccuracies associated with reading from a user interface. Saving a float value seems like a good solution.

Regarding improvement # 2:
Don't forget to provide specific information about the setup of the testing mast, detailed enough that someone could recreate your setup without speaking to you. You provide good detail here, but make sure it gets saved systematically somewhere. We can talk about the best format for this. I think it's a good idea to test whether the setup is interfering with itself, but we need to approach this systematically.

Regarding improvement # 3:
This has to be an error. Does the issue remain if you re-order the units? I can't see a reason why the signals would be getting progressively weaker as you go down the array; each SDR has its own antenna with its own preamp. I'd definitely bring this up with Igor.

russellmsilva, 2018-09-27:
A video overview of the testing conducted 9/23/2018 can be found [here](https://www.youtube.com/watch?v=DcbEhXReK4k).  Please bear with the video quality/stability (it was difficult to record a video of the generated phase differences in real-time when holding the transmitter and my laptop).

Originally, the testing was performed with the sole purpose of video documenting the technical issues of our phase interferometry system. This was done so that experts who were able to implement similar direction-finding systems could help us debug these issues. However, the testing resulted in unexpected observations that necessitate further experimentation to isolate what the specific technical issues are.

Testing was conducted outdoors on the baseball field. I placed the transmitter at distances perpendicular to the antenna array (90 degrees) to decrease the testing parameters to tag-to-array distance, movement, transmitter gain, and receiver gain. Notable findings include:

1. Based on the transmitter and receiver gains, there was a distance from the antenna array at which the received signal would dissipate lower than a visible amplitude on the waveform display in GNU Radio. This is a beneficial quality of the testing environment, because the transmitted signal can be attenuated to a negligible amount by the time it reaches buildings/objects near the baseball field with low transmitter gains. The receiver gain can also be lowered to an amount to which reflected signals from nearby buildings/objects would not be detected by the receivers. Due to the brevity of testing time available, the exact transmitter gain/distance/received amplitude relationship of the CC1310 could not be determined.

2. At a ⁓35-meter distance from the antenna array, the phase difference had a stable measurement close to 0 degrees when crouching down. When standing up with my body near the transmitter, the measured phase difference fluctuated to a different value. Also, the phase difference varied more than expected when the transmitter pole was tilted to the left or right from its vertical position.

Because of this second finding, I am going to modify the transmitter setup so that I can stay away from the CC1310 during testing and to more effectively record the system’s performance. The modifications will include building a base for the transmitter pole and programming the CC1310 to send an unmodulated 433.92 mHz signal without a connection to my laptop. These modifications will be done so that the system can be debugged and improved incrementally, adding interfering objects once our system can generate accurate AoAs without interfering objects (such as myself).